BondtechAB/Bondtech_Prusa_i3 #3: MMU2 version of MK3S filament sensor mod?

monglet, 2019-03-01:
Any chance of a top part for the latest update that incorporates the fixing for the festo coupler?

Pitterz, 2019-03-04:
X2.  It looks like an easy fix but the print would involve supports :(

Dutchy-79, 2019-03-04:
Depends on how you draw the part, make it bulky and you don't. https://www.thingiverse.com/thing:3461572
https://www.thingiverse.com/thing:3463618

ruedli, 2019-03-14:
when support for the MMU2S is needed, the detector needs to be triggered by door movement, which involves something totally different for the optical sensor part in addition to attaching the FESTO coupler to the extruder.

The Prusa MK3S / MMU2 compatible parts are here:
https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/adapter-printer-mmu2s.stl

https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/fs-cover-mmu2s.stl

and

https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/extruder-idler-mmu2s.stl

Notice the lever on the idler needed to transfer the movement towards the IR filament detector !