BondtechAB/Bondtech_Prusa_i3 #6: Temperature Issue on mk3s with Mosquito

rcorgill, 2019-10-05:
Hopefully this is the right place to post this...I recently installed the Bondtech\Mosquito kit on my Mk3s along with slice engineering heater cartridge and slice engineering thermistor.  I flashed the 3.8.0 file named "bondtech_fw_MK3s-16_SE_HT-Thermistor_3_8_0.zip" and the firmware flashed successfully.  However I cannot print higher than 295 degrees....I confirmed with Bondtech support that the firmware (with HT-Thermistor in the file name) should have the correct tables to support the high temp thermistor but that doesn't look to be the case.   Most guides online for compiling the firmware are for the factory hotend\extruder so I'm not sure what to do...Any ideas?  Thanks

Tooke, 2019-12-09:
Confirmed the same issue with the same setup as above.

bondeus, 2019-12-12:
We also made versions to support Slice Engineering high temperature thermistor. We reduced the MINTEMP threshold to allow use in colder environments. The max temp was not changed. If you are an experienced and knowledgeable user you can change the HEATER_0_MAXTEMP parameter to fit your needs. 

Pasted from: https://www.bondtech.se/en/knowledge-base/firmware-updates-for-prusa-i3/