BondtechAB/Bondtech_Prusa_i3 #9: Prusa Firmware as Fork

3d-gussner, 2019-11-12:
I am glad to see that you provide the Prusa Firmware modifications needed for the BMG for Prusa i3.
It would be way easier for us if you could get a github fork from https://github.com/prusa3d/Prusa-Firmware and make the changes commits in your fork.
Now we have to read the commits, download source.zip file, unpack it and compare manually what changed. With a fork we could just compare the changes via github, here an example from my fork:

3d-gussner, 2020-05-20:
@lejo1512 I really like to see that Bondtech has now a fork of Prusa-Firmware and publishes the sources code. But please open the issue section so we can comment also in the new repo https://github.com/BondtechAB/Bondtech-Prusa-Firmware

lejo1512, 2020-05-21:
@3d-gussner Thanks for the reminder, should be open now.

3d-gussner, 2020-05-21:
Great, close this issue

closed 2020-05-21 by 3d-gussner