BondtechAB/Bondtech_Prusa_i3 #11: Where shall I place the MK3s "#elif define" changes?

elektronchika, 2020-02-04:
Dear Bondtech team,

thank you providing all the changes that you have implemented in Prusa firmware. I only cannot understand where I have to place the last 3 changes for MK3s firmware? I would be very appreciative if you let me know.

Thank you and kind regards,
Lazar

lejo1512, 2020-02-04:
These changes are made in mmu.ccp during "mmu_load_to_nozzle()"and  "can_load()".

dmnkhhn, 2020-05-19:
Hello!

I have to make the changes from the "MK3s Mosquito with high temperature thermistor (SliceE_HT)" part of the documentation in the mmu.cpp of firmware 3.9.0 but I can't find where to put them.

Those lines are:
```
#elif defined(BONDTECH_MOSQUITO) current_position[E_AXIS] += 70; //Bondtech_Mosquito 70mm from drive gear to melt zone
#elif defined(BONDTECH_MOSQUITO) current_position[E_AXIS] -= 62; // Pull back 62mm, 8 mm below drive gear
#elif defined(BONDTECH_MOSQUITO) current_position[E_AXIS] += 23.4f; //Bondtech Mosquito 9 mm longer t melt zone
#ifdef BONDTECH_MOSQUITO current_position[E_AXIS] += 16.4f; //2mm further through Mosquito heat block
```

Thanks for any help. :-)

3d-gussner, 2020-05-19:
Checkout https://github.com/Zaribo/Zaribo-Firmware there you can compare it with Prusa and see the changes you need to make.
Or you compile this firmware.