BondtechAB/Bondtech_Prusa_i3 #12: MK2.5s Mosquito with high temperature thermistor (SliceE_HT) firmware not showing right temperature

RiotShooter75, 2020-04-10:
I flashed this firmware to my Prusa and my printer is not showing the correct temperature. Thouhgts?

jgrg1, 2020-04-19:
Hi, I have exactly the same issue, exactly the same build (MK2.5S with Mosquito hotend and SE HT thermistor). I've dug through the firmware codebase, and I believe that it's using the correct table in thermistortables.h with TEMP_SENSOR_0 = 800 from the variants\1_75mm_MK25S_Bondtech-RAMBo13a-SliceE_HT-Mosquito.h file.

Is there a solution to this?

RiotShooter75, 2020-04-19:
Apparently Prusa broke something in a recent update/the rambo used on the MK2 variants has issues with the Slice Thermistor. I had to move back to using an E3D thermistor on my Prusa.

3d-gussner, 2020-04-19:
Please checkout vertigo235 fork of prusa firmware... I think he got it working.
Other option is zaribo github fork.

Panayiotis-git, 2020-04-19:
It seems like the build for the mk25s has not used the definition TEMP_SENSOR_0 = 800 but the TEMP_SENSOR_0 = 5.