BondtechAB/Bondtech_Prusa_i3 #27: After mounting, X endstop is not touching

nbouayad, 2023-09-17:
Hello,
On a Prusa MK2,the extruder rear cables prevent the x-endstop to be pushed, as seen in the attached pictures :

[image]

[image]

[image]
 
What can be done??!
Thanks,

lejo1512, 2023-09-18:
I assume you are using "X-Carriage-modded(R2)__for_MK2(s)" with the extended arm to hit the switch?
Is the 

With the R3 cable-holder design the sleeve and zip-ties needs to be properly tightened around the cables.
The cable should be within the geometry of the cable-holder. 

You could try the MK2.5 version of the cable holder, gives some more space around the frame: 
https://github.com/prusa3d/Original-Prusa-i3/blob/MK2.5/Printed-Parts/stl/cable_holder.stl

[image]
(From Prusa MK2.5 assembly guide)

nbouayad-wincomplm, 2023-09-18:
@lejo1512, thank you for your answer.
Yes, I am using the modded R2 for MK2(s).
I will try the MK2.5 version of the cable holder and come back to you.
Kind regards,

nbouayad, 2023-09-20:
@lejo1512, using the MK2.5 cable holder solved the situation!!
Thanks again!

closed 2023-09-20 by nbouayad